emoncms/group #12: Unable to make dashboards after using the group module

LexRillo, 2017-08-24:
I cannot create dashboards for a conflict between the Modules/graph/graph_model and Modules/graph/widget/graph_widget since the __construct method of Graph asks for mysqli and group and widget only uses mysqli. I have attached an image with the error print.
[image]

cagabi, 2017-08-29:
Hi Alejandro,

I am back now, catching up with things. I'll have a look into this soon.

  Carlos

cagabi, 2017-08-31:
Hi, this is fixed now.
But I have just found a new bug #13, so you may want to wait to update the graph module until that gets fixed

LexRillo, 2017-09-04:
The message has changed and the board is visible but now I get:
[image]

cagabi, 2017-09-04:
Hi, it looks like you haven't updated the graph module. Do it from the development branch: [graph module](https://github.com/carboncoop/graph/tree/develop)

closed 2017-09-06 by cagabi