emoncms/group #15: Administrators able to eliminate other administrators

LexRillo, 2017-09-19:
On the current implementation, administrators of the group have the capability of removing another administrators which could be a problem for users that have huge amounts of data on the feeds and get eliminated by mistake or by malice.

cagabi, 2017-09-20:
You are right. There is a functionality I'd like to implement which is "rights over a user's data". This means that a user could choose if and administrator have _full access_ (create and delete feeds or even delete the user) or only _read access_.

I didn't implement it yet as the original specification was aimed to manage users so it made sense to have always have _full access_. I think I'll implement it soon though

LexRillo, 2017-09-20:
Thanks, cagabi. I know we talked about this but I just got reminded of it.
So no problemo.