emoncms/group #19: When user has "member" role in a group, he/she looses it's admin rights for other groups in Graph module

cagabi, 2017-12-06:
From Ella's email:

> I think a clear problem, as you state, is with the “member” role.  **It does not display group graphs for the group in which you are a member, nor can you view any graphs in other groups, even if you are an administrator or sub-administrator**.  So if the “member” role is selected for any group it is messing up the visualisation of group graphs in all groups for that account, not just that particular group in which you are a member.

EllaQuigley, 2017-12-07:
Thanks Carlos, 

I'm glad that you have managed to replicate the same bug at your end.

Ella

cagabi, 2017-12-15:
Fixed. 
Please update Group module

EllaQuigley, 2017-12-18:
Carlos,

This is still not working.  Being a member of one group still means that you cannot create group graphs in another group, even when you are administrator of the other group.

Ella

cagabi, 2017-12-19:
Can this be related to the other error you mentioned the other day about the Save/Delete buttons not showing? 
This error was about graphs not showing

EllaQuigley, 2017-12-19:
Hi Carlos,

This problem with the save/delete buttons is now fixed, so they appear to be unrelated.

Thanks

cagabi, 2018-03-06:
Ella,

I have looked at this issue again and found the problem. Sorry I was not able to replicate it, you explained the problem well and only now I have seen it clear.

Basically, the concept of "member" role is not possible to implement in the Graphs module.

On top of that, after our phone conversation ages ago I think this role has to be rethought to fit in new case scenarios.

So I think the best option for now is to remove the "member" role (you would also have to change the role of your users that have it). The problem will not be fixed but will have to be revised once the new definition for "member" is done.

Please let me know what you think before I close this issue.

All the best,

    Carlos

EllaQuigley, 2018-03-08:
Hi Carlos,

Thanks for your email.  I’m glad you have now seen the problem.

Given that the member role does not work, and I don’t know why it doesn’t work, I think it would be helpful to have a quick chat with you about what the problem is, so I can decide if and how we work around it.

Am I right in thinking that the only difference between the member role and the sub-administrator role was that members can’t see who else is in the group whereas the sub-administrators could?  Or were there some other differences?

Kind regards,
Ella

cagabi, 2018-03-08:
Hi

> Am I right in thinking that the only difference between the member role and the sub-administrator role was that members can’t see who else is in the group whereas the sub-administrators could?  Or were there some other differences?

Yes, that was the idea: the member would not know who else is in the group but would be able to see the group graphs. There is a conceptual contradiction there, in a graph where you are showing user's feeds it is not possible for the member to have access to those feeds without knowing who they belong to.

My original plan for this member role was that group admins would have the capacity to create group feeds (eg, for aggregation) and the members would have access to them for example in the graphs module. This would fit in an older Carbon Co-op and openenergymonitor project: the [virtual microgrid dashboard](https://openenergymonitor.org/dev/smartgrid/). I thought this was an opportunity to add the infrastructure (member role) that would make it possible. But as it was not included in the specification it wouldn't have been fair I spent any time on it so left it there but not working.

I am happy to talk if it helps

cagabi, 2018-03-20:
Finally we agree this "member" role doesn't work conceptually. It will be further developed in the future according to a new specification

closed 2018-03-20 by cagabi