emoncms/group #20: Add bootstrap-switch to local lib

cagabi, 2017-12-08:
Depending in fetching it from a public repository will make the Graph module in a local emonCMS installation not to work properly if the internet connection fails

EllaQuigley, 2017-12-11:
Hi Carlos,

Yes, you are correct, we had internet problems last week, which explains why some of the buttons were missing.  It would be really useful if you could add the component so we are not dependent on the internet.

Regards,
Ella

cagabi, 2017-12-13:
Ups, I already did this and didn't remember. [Commit: Mode switcher files in /Lib](https://github.com/carboncoop/graph/commit/109ea72caea98efd838a3282ce84303f22be1885)
You need to update the Graph module
  Carlos

closed 2017-12-13 by cagabi

EllaQuigley, 2017-12-13:
Great, thanks Carlos!