emoncms/group #26: How does CSV download work in feed view?

cagabi, 2017-12-20:
When interval for the csv is higher than the feed interval it seems to only download the closest datapoint to  the specific time not an average of all the datapoints included in the interval. Is that right?

EllaQuigley, 2018-01-25:
I've done some checks on this.  There is no data averaging irrespective of where you download data from (e.g. graphs or feeds).  Also, it doesn't download the closest datapoint, if there is not a reading at that exact time then you get no data at that time, so if you want hourly data there has to be an actual log on the hour: hh:00, if you want it daily there has to be a log exactly at midnight 00:00 and so on.

cagabi, 2018-01-26:
Thanks for looking into this.

To be honest, this csv download seems pretty useless unless you do it with the right interval.. I need to think what to do

Are you sure, the average option doesn't work in the graph view?, did you tick the Average checkbox?