emoncms/group #30: Groups module display problem in Internet Explorer

EllaQuigley, 2018-01-17:
When in internet explorer browser, on the groups page, the groups appear on the left hand side, but when you click on them the group information does not load.

cagabi, 2018-03-13:
Hi Ella,

Which version of Internet Explorer are you using?

Doing a search about emonCMS and IE compatibility I came across this: https://github.com/emoncms/emoncms/blob/master/docs/WindowsInstall.md

Where it says

> Internet explorer 9 - works well with compatibility mode turned off. F12 Development tools -> browser mode: IE9. Some widgets such as the hot water cylinder do load later than the dial.
IE 8, 7 - not recommended, widgets and dashboard editor do not work due to no html5 canvas fix implemented but visualisations do work as these have a fix applied.

cagabi, 2018-03-14:
Closing this issue after phone conversation as IE not required

closed 2018-03-14 by cagabi