emoncms/group #31: Dashboard option/tab has disappeared

EllaQuigley, 2018-01-24:
The dashboard option is now missing from the dropdown menu.  It was there last time I used emoncms last week, nobody else had used it during this period and Alex never made any changes to the files, so I am quite confused!
I'll email you a screenshot as can't see if I can do it here.

Thanks,
Ella

cagabi, 2018-01-25:
Hi Ella,
If the module is there the menu should be showing.

So just in case, can you confirm that when you go to
                   yourdomain/emoncms/dashboard/list
your dashboards are still there.

EllaQuigley, 2018-01-25:
Hi Carlos, 

I guess it is not there, this is the message I get on the screen:

URI not acceptable. No controller 'dashboard'. (list/)

So have we deleted the module?

cagabi, 2018-01-25:
It looks like it's been deleted.
I'll work on this with Alex

EllaQuigley, 2018-01-25:
Hmmm! Ok, thanks!!

Maybe it wasn't there last week and I just didn't notice.

cagabi, 2018-01-25:
Alex has reinstalled the Dashboard Module. So you should be able to see it again

closed 2018-01-25 by cagabi