emoncms/group #35: Active feeds column in groups view should colour feed number by status

beaylott, 2018-05-17:
In the active feeds column (see picture) feeds appear to be coloured by whether they have received any data or not. I have observed that where a feed has stopped recording values (because device has stopped posting or otherwise) the number still appears as green. It would be good if the colour reflected whether or not the feed has updated within a specific time period or not e.g.

green = all feeds have updated within last 1 minutes
amber = all feeds have updated within last 10 minutes
red = no feeds have updated within last 1 hour

[image]

cagabi, 2018-05-21:
Hi,

This is how it works in the groups view: 
- if a feed has been upadted in the last 48h then  it's active 
- 3/5 means that 3 feed have been updated in the last 48h
- the color is set depending on the proportion of active feeds of the total of feeds

I have had a look at the color coding in the feeds view:
 - green: updated time < 25s
- amber: t < 60s
- orange: t < 2h
- red: t >2h

I agree we need to change it in the groups module. So I'll do it as you say but keeping the same timing than the feeds view.

cagabi, 2018-05-21:
I have changed it, now it shows how many feeds have been updated and when (to see the meaning of the colors hover over the question mark in the table header)

I have to say it's quite full on with so many colors and in fact not that clear. 

I am closing this issue now but feel free to open it again if you  don't like those many colors

closed 2018-05-21 by cagabi