emoncms/group #37: Unable to add users to new groups on emoncms.org

pb66, 2018-06-06:
On emoncms.org I am unable to add a user to a new group as reported on the OEM forum in the ["One device - different users"](https://community.openenergymonitor.org/t/one-device-different-users/7541/6?u=pb66) thread. Just tried it again and it is still a current issue, although oddly, I have noticed that I can add the same user to a previously created test group that I created way back when the groups module first appeared on emoncms.org. So I am currently able to add a user to an old existing group, but not to a newly created group. 

[image]

cagabi, 2018-06-07:
Hi Paul,
I am testing in my self hosted up to date installation and everything is working ok. I am waiting for my email verification in emoncms.org to try it there but I am afraid I cannot help on that one.

cagabi, 2018-06-07:
Ah, I get all notifications for issues in this repository :)

cagabi, 2018-06-07:
Hi again, tried in emoncms.org and you are right it's broken. I got the same pop ups, first saying "undefined" then saying "Name already exists" and now the page loads but the group seems to be empty

pb66, 2018-06-07:
So should this be moved to the emoncmsorg repo? or will @TrystanLea  (or @emrysr) pick this up here ?

We don't even know if emoncmsorg is running the latest group module or if it's modded in anyway for emoncmsorg specifically.

cagabi, 2018-06-07:
Good point. Let's wait and see if we hear from them soon and if not open an issue in emoncms.org