emoncms/group #42: When impersonating another user, display current user as well as original login

takkaria, 2018-06-29:
At the moment when you impersonate another user, it says "You were originally logged in as <username>."  It would be useful if it said "Currently logged in as <username>" as well.

pb66, 2018-06-29:
I noticed this too.

I also noticed that the current warning in the navbar takes up a huge amount of space that was already tight if you have several dashboards to link to.

I would like to see the far right "login/log out" button re-purposed to show who is logged like most online accounts seem to, and when clicked it drops down to show links to account page, groups? and then a "log out" option at the bottom.

Perhaps the user name/gravitar background could turn red if logged in as someone else?

I also think the current account "name" should always be displayed (I have it on the far left of the nav bar) so that when you have several accounts open to view (public pages or using apikeys) you always know what account you are looking at, it also doubles as a home button to take you back to the "main" dashboard.

cagabi, 2018-07-04:
@pb66 I think those are very good ideas about the login/log out button. Should that go as an issue in the meoncms repo?

cagabi, 2018-07-10:
Done
I have changed the text to show the original user and the new one in this form 
`original_user => new_user`

closed 2018-07-10 by cagabi