emoncms/group #45: You have to know a user's password to add them to a group

takkaria, 2018-07-26:
This has been unhelpful in setting up groups for Nobel Grid - I wanted to set up groups for the different sites we have installations at and I can't.

cagabi, 2018-07-27:
I see your point from the perspective of us being the only ones managing the whole emoncms installation. But on the other hand it makes sense that you cannot add users to a group (and have access to their accounts) without some kind of consent.
Without requiring a password a user from Lancaster could (knowing any username) create a group, add the users and access their accounts

takkaria, 2018-07-27:
I understand your point.  I wonder if there is a way we can set things up so both situations can be accomodated?  

I think the security model at the moment is the thing that holds this back - it seems to me that being able to create groups or log in as other users should have some kind of permission control that isn't about knowing other people's passwords or usernames.  I'm imagining some kind of least privilege-style capabilities system like e.g. WordPress/Civi/AWS use, so that you can only do these things if you are explicitly granted the ability to do them.  Then you'd have a capability like 'Can add members to groups (passwordless)' as well as perhaps 'Can add members to groups (with password)'.  I guess that's what I was wondering about in #46 as well.