emoncms/group #46: Disallow users from creating/editing groups

takkaria, 2018-07-26:
For the users we're creating for Nobel Grid, they have access to the groups UI and can create groups.  It would be good to be able to disable this.

takkaria, 2018-07-27:
Maybe Emon itself needs some kind of permissions system, so that you can grant users different capabilities?

cagabi, 2018-07-27:
You are right that there is no control access in emonCMS. Basically you can be superadmin ("admin" column in users table) or normal user. This is because until we made the Group module the use for emnCMS has always been single users accessing their own data.
In terms of your point, I don't see the problem of users having access specifically to the Group module. It's a bit silly to have it there if they are not going to use it, but they also have access to all the other modules (input, feed, device) and we don't want the users to play with them. So I agree with you that a permissions system would be good.

cagabi, 2018-07-27:
Thinking about it from a different angle: what is what we want for Nobel Grid users? Only access to the dashboards? Could there be a "Dashboard only" type of login?

takkaria, 2018-07-27:
Yeah, I suppose I am also thinking that we need to disable access to a bunch of things for managed users.  For user experience - so they don't have access to stuff that doesn't make sense for them / is useless - and also for privacy/data access reasons - if they change their device details for example they might be able to read other people's data.  (Though I know Ben has written safeguards for this.)

cagabi, 2018-07-27:
Oh interesting, you need to tell me how they can do that ;)
I guess the "only dashboard login" is a very specific solution while adding some kind of permissions system has a great potential

takkaria, 2018-07-31:
I am going to spend some time this week trying to prototype a capability system.  I'll let you know how I get on!

cagabi, 2018-08-01:
Cool :)