emoncms/group #47: Respond to back button

takkaria, 2018-07-31:
When you use the back/forward buttons on the group page, nothing happens.  It would be good to use the [History API](https://developer.mozilla.org/en-US/docs/Web/API/History_API) so that it works.

takkaria, 2018-07-31:
I was thinking about adding something to respond to the `[hashchange](https://developer.mozilla.org/en-US/docs/Web/Events/hashchange)` event.  This would mean that when you press back the group changes back to the old group.  I have used this on other projects, would be happy to implement it here too.

cagabi, 2018-08-01:
I see, in fact hashchange is what we use in MyHomeEnergyPlanner.
Great, I am very pleased you are on this too!