emoncms/group #48: Button to reset password + send login details to user

takkaria, 2018-07-31:
At the moment it is a manual process to send users their password if they are already created (e.g. using the script as has been used for Nobel Grid).  It would be good to be able to send passwords when editing a user, using some kind of editable template if possible.

cagabi, 2018-08-02:
I am working on this now, so far the wildcards I have added are {{password}} and {{username}}.
Shall I add "Name" to the edit user modal and the {{name}} wildcard?

cagabi, 2018-08-02:
I have done it:
 - Added 'Name' to the "Create user and add to group modal"
 - Added the send email feature to the edit user modal
 - Added inputs for email subject and body
 - Wildcards to use: {{name}}, {{username}}, {{password}}, {{role}}, {{appname}} (set in settings.php) and {{path}} (useful for making links. Hover the question marks on the modals for clarity

cagabi, 2018-08-02:
Testing it in the server there seems to not be working

cagabi, 2018-08-02:
Sorted

closed 2018-08-02 by cagabi

takkaria, 2018-08-06:
Great, thanks!