emoncms/group #55: Error: parsererror: SyntaxError: Unexpected token < in JSON at position 0

glynhudson, 2019-04-27:
When testing darksidebar PR https://github.com/emoncms/group/pull/54 I get error `Error: parsererror: SyntaxError: Unexpected token < in JSON at position 0`. This is the first time I've installed and tried to use the group module therfore I'm unsure if this error is caused the darksidebar changes or not?

[image]

emrysr, 2019-04-29:
fixed in commit of emrysr/emoncms

https://github.com/emrysr/emoncms/commit/6b3cc70a38c5790c526d67730b8d320a01471c88

glynhudson, 2019-04-30:
I'm still getting the same JSON error after updating this PR. Is that the commit you intended to mention above ^? That commit is related to fixing the bookmark issue on Emoncms core.

closed 2019-12-13 by TrystanLea