emoncms/group #57: LICENSE

reedy, 2019-07-02:
What's the license for this repo?

Is it AGPL-3.0-or-later as per https://github.com/emoncms/emoncms ?

cagabi, 2019-07-03:
Not sure about the "later"
The only reference to licnse is what we added on the beginning of the files (like many other emonCMS files"

> All Emoncms code is released under the GNU Affero General Public License.

reedy, 2019-07-03:
https://github.com/emoncms/emoncms/blob/master/COPYRIGHT.txt#L6-L9 says otherwise (at least for the main/central/whatever repo)

```
    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU Affero General Public License as
    published by the Free Software Foundation, either version 3 of the
    License, or any later version.
```

cagabi, 2019-07-04:
Do you think this module nee to have a similar copyright?

reedy, 2019-07-04:
The module needs a copyright, yes. I'm not so bothered about what it is (IANAL), for ease, it's easier if it's the same as the parent project

I believe this counts as a derivative of https://github.com/emoncms/enomcms but don't quote me on that 

>We believe open source is a better way of doing things. Emoncms is available under the GPL Affero licence (AGPL).

So, as per https://help.github.com/en/articles/licensing-a-repository

>For your repository to truly be open source, you'll need to license it so that others are free to use, change, and distribute the software.

:)

cagabi, 2019-07-18:
I have added a copyright and copied the license. I think it should be now, have a look.
Thanks

commit 5d2d2890 (2019-07-18): Fixes #57: missing license
closed 2019-07-18 by cagabi