emoncms/group #61: Downloadlimit settings not being fetched - browser error

jteez, 2019-10-06:
Hi Team,
I don't know if this is the best place, but I've been battling for a while to trace an issue on my brand new installation. It's running all with latest (stable branches if available)
Version: 10.1.8

The main install seems to work but navigating to the group module using the menu on the left, there was a notification saying 
    unexpected token ';'
    route:group
    line 1421.
I was getting no errors in log even with it set to information
I adjusted the logging settings and got it to show in a bit more detail in chrome debugger
Turns out there was a line being interpreted 'var downloadlimit =;'
searching for that, I found a line in group_view.php & various settings in settings.ini & php
I confess to not knowing how the various settings.xxx (ini/php) and default-settings.xxx files are used... so not sure if I've actually fixed this in the best place but I've fixed the issue by replacing the following lines (appears twice)
```
     global $feed_settings;
     echo $feed_settings['csvdownloadlimit_mb'];
```
TO
```
     global $settings;
     echo $settings['feeds']['csvdownloadlimit_mb'];
```

No idea if that's the right thing to do or not but it's working now so thought I'd drop you all a note.
Thanks,

TrystanLea, 2019-12-13:
Thanks @jteez this is now fixed and a number of other items that where broken, sorry for the delay

closed 2019-12-13 by TrystanLea

jteez, 2019-12-13:
Great stuff ,thank you!